Toylerrr/Flashforge-for-Cura #2: Problem with octoprint and Cura with duel filament prints

Toylerrr, 2019-07-27:
Causing the head to crash the wrong way and not hitting the end stop. maiking it sound like the printer is trying to explode.

valueerrorx, 2019-08-31:
eerm... did you fix that already?  

did you have a look at those 2 files for flashforge creator pro and dreamer3d ?

those have a little bit different gcode

valueerrorx, 2019-08-31:
these files are from Simplify3D which supports flashforge printers

[FlashForge Creator Pro.fff.txt](https://github.com/Toylerrr/Flashforge-for-Cura/files/3562399/FlashForge.Creator.Pro.fff.txt)
[FlashForge Dreamer.fff.txt](https://github.com/Toylerrr/Flashforge-for-Cura/files/3562400/FlashForge.Dreamer.fff.txt)

Toylerrr, 2019-08-31:
I'll talk a look at those and see what I can do thanks for the references

Meins321, 2019-09-17:
Would be nice if you update the readme page with date and maybe the cura version you are using?

Currently Cura 4.2.1 works with single extrusion with this Profile but not Dual..:
https://www.thingiverse.com/thing:3569113/files

The Extruder runs to the sides and waits very long to heat up every time.. would be nice to print in one flow!

Toylerrr, 2020-01-12:
Fixed in newest version

closed 2020-01-12 by Toylerrr