Toylerrr/Flashforge-for-Cura #5: flashforge dreamer3d

valueerrorx, 2020-01-14:
do you have any idea how to adopt this cura profile for the dreamer3d ? 
i managed to get it to work with one extruder but it did a really bad printjob and used to much filament..

it works gread with slic3r  https://github.com/valueerrorx/slic3r-dreamer3d
so the gcode shouldn't be the problem.  the same gcode used in cura doesn't work at all... 

i think it's a problem with the flavor

valueerrorx, 2020-01-14:
https://github.com/valueerrorx/Flashforge-for-Cura
here is the (half baked) version for dreamer3d (a fork of your repository) with all of the neccesary files for dreamer3d as a starting point for everyone who want's to give it a try

Toylerrr, 2020-01-16:
I dont have a dreamer but after looking some stuff up i have some info that might help. 

The Dreamer uses gcode not x3g so it should be easy to get it working. 

This is the gcode straight from flashprint for the starting Gcode
```
M118 X17.70 Y17.70 Z34.89 T0
M107
G90
G28
M132 X Y Z A B
G1 Z50.000 F420
G161 X Y F3300
M7 T0
M6 T0
M651
M907 X100 Y100 Z40 A100 B20
```
This is what it had from the End Gcode
```
M104 S0 T0
M140 S0 T0
G162 Z
G28 X Y
M132 X Y A B
M652
G91
M18
```

Also in your cura setup you have the dreamer set up with 
machine_x3g_variant": "r1d",  **(You shouldnt need this)**

"file_formats": "application/x3g",  **(this should be something like "file_formats": "text/x-gcode")**
and

"machine_gcode_flavor": { "default_value": "RepRap" }, **(should be something like "machine_gcode_flavor": { "default_value": "Makerbot" } )**

valueerrorx, 2020-01-19:
wow thx.... i'm going to give it a spin tomorrow

valueerrorx, 2020-01-21:
holy shit .. this went south.. 
first of all the flashforge dreamer3d claims that there is an error in the file when using "Makerbot" flavor..  i already figured out that it would accept the file when using "UltiGCode"  -- so i did that..

then the dreamer started to move down the plate (but didn't stop) and immediately started printing (moving the extruder printing into thin air) but it didn't pre-heat..  lucky me ;-)

so i changed the gstartcode and endcode to what i know is working in "slic3r" 
and voila.. it heats up and perfectly starts printing..  well..  almost..  haha..  it heats up both extruders even though only one is needed and it spills the filament and moves in it's own mess..

[image]

do you have an idea how to fix this?   
it heats up to 220°C which i thought is to high but then slic3r does the same thing..

valueerrorx, 2020-01-21:
start_gcode = 
6
```
7
M140 S55;   Heat bed up to first layer temperature
8
M104 S220 T0;   Set nozzle temperature of Right extruder to first layer temperature
9
M104 S220 T1;   Set nozzle temperature of Left extruder to first layer temperature
10
M107;   Fan off
11
G90;   Absolute Programming
12
G28;   Move to Home position 
13
M132 X Y Z A B;   Load current home position from EEPROM
14
G1 Z50 F600;   Move bed down to allow safe movement of nozzles lo left front
15
G1 X-110.5 Y-74 F6000;   Move nozzles to left front
16
M7;   Wait For Platform to reach target temperature
17
M6 T0;   Wait For Right nozzle to reach target temperature
18
M6 T1;   Wait For Left nozzle to reach target temperature
19
M907 X100 Y100 Z40 A100 B100;   Set digital potentiometer value. A and B typically 100 for tough filament like ABS and 80 for brittle like PLA. A is Right B is Left.
20
G1 Z0.6 F3300; Go to start height
21
G4 P2000;   Dwell time
22
G1 Z0.2 F7200.000;   Move to first layer height
23
M108 T0;   Tool change to current extruder
24
```
25
​
26
​
27
end_gcode =
28
```
29
M107;   Disable cooling fan\n
30
M104 S0 T0;  Set Nozzle temperature of Right nozzle to zero
31
M104 S0 T1;  Set Nozzle temperature of Left nozzle to zero
32
M140 S0;   Turn bed heating off
33
G162 Z;    Home positive for Z axis
34
G28 X0 Y0;   Return X and Y to home position
35
M132 X Y Z A B
36
G91;   Set to relative positioning
37
M18;   Disable stepper motors for all axes
38
```

valueerrorx, 2020-01-21:
start_gcode = 
```
M140 S55;   Heat bed up to first layer temperature
M104 S220 T0;   Set nozzle temperature of Right extruder to first layer temperature
M104 S220 T1;   Set nozzle temperature of Left extruder to first layer temperature
M107;   Fan off
G90;   Absolute Programming
G28;   Move to Home position 
M132 X Y Z A B;   Load current home position from EEPROM
G1 Z50 F600;   Move bed down to allow safe movement of nozzles lo left front
G1 X-110.5 Y-74 F6000;   Move nozzles to left front
M7;   Wait For Platform to reach target temperature
M6 T0;   Wait For Right nozzle to reach target temperature
M6 T1;   Wait For Left nozzle to reach target temperature
M907 X100 Y100 Z40 A100 B100;   Set digital potentiometer value. A and B typically 100 for tough filament like ABS and 80 for brittle like PLA. A is Right B is Left.
G1 Z0.6 F3300; Go to start height
G4 P2000;   Dwell time
G1 Z0.2 F7200.000;   Move to first layer height
M108 T0;   Tool change to current extruder
```

end_gcode =
```
M107;   Disable cooling fan
M104 S0 T0;  Set Nozzle temperature of Right nozzle to zero
M104 S0 T1;  Set Nozzle temperature of Left nozzle to zero
M140 S0;   Turn bed heating off
G162 Z;    Home positive for Z axis
G28 X0 Y0;   Return X and Y to home position
M132 X Y Z A B
G91;   Set to relative positioning
M18;   Disable stepper motors for all axes
```

valueerrorx, 2020-01-21:
from looking at the produced gcode files (ultiGcode vs makerbot) i can say that the ultigcode (the one that works) moves the filament almost twice as much..  (the E numbers)

this has to be the reason.. hmm.. what now...
[image]

valueerrorx, 2020-01-22:
okay.. i found a manual fix and therefore the reason why "flashforge dreamer3d" cancels the print immediately with the makerbot flavor (which is the correct one)

[image]

those 5 lines are the cause..  

T0
M190 S50
M104 S200
M104 S200
M116

if i delete them in a text editor the dreamer3d starts printing and does a very good job at it..

this selects extruder 0, preheats bed an nozzle afaik..  but it's also part of the gstartcode i provided so it makes no sense that those lines are added..  

with a LOT of trial an error i figured out that only one of them causes the print to cancel imediately
T0

i have to remove T0 and the file works fine..

Toylerrr, 2020-03-08:
Those lines are added by Cura itself, Sounds like for it to not have those you would have to work the gcode start script to have the correct temp variables.

M190 S{material_bed_temperature} 
M104 S{material_print_temperature}

Toylerrr, 2020-03-08:
```
M140 S{material_bed_temperature};   Heat bed up to first layer temperature
M104 S{material_print_temperature} T0;   Set nozzle temperature of Right extruder to first layer temperature
M104 S{material_print_temperature} T1;   Set nozzle temperature of Left extruder to first layer temperature
M107;   Fan off
G90;   Absolute Programming
G28;   Move to Home position 
M132 X Y Z A B;   Load current home position from EEPROM
G1 Z50 F600;   Move bed down to allow safe movement of nozzles lo left front
G1 X-110.5 Y-74 F6000;   Move nozzles to left front
M7;   Wait For Platform to reach target temperature
M6 T0;   Wait For Right nozzle to reach target temperature
M6 T1;   Wait For Left nozzle to reach target temperature
M907 X100 Y100 Z40 A100 B100;   Set digital potentiometer value. A and B typically 100 for tough filament like ABS and 80 for brittle like PLA. A is Right B is Left.
G1 Z0.6 F3300; Go to start height
G4 P2000;   Dwell time
G1 Z0.2 F7200.000;   Move to first layer height
M108 T0;   Tool change to current extruder
```

This would only be for duel prints because it heats up both sides of the extruder would heat up.

closed 2020-04-19 by Toylerrr

KeltE, 2021-03-19:
Hello! I don't have a Dreamer but a DreamerNX and I made a definition for Cura. I looked at your definition. When changing printheads, maybe adding this line to the extruder file helps? 
"machine_extruder_start_code": {
             "default_value": "\ nM108 T0; Select Active Extruder \ n"
           }
And the beginning of the G-code maybe should
 \nM104 S{material_print_temperature_layer_0} T{extruder_nr} ;Set nozzle temperature to first layer temperature\n

I also couldn't get rid of the T0 command before the start code.