Toylerrr/Flashforge-for-Cura #6: Cura Profile for FlashForge Creator Pro - Extruder 1 Not feeding properly

CrouchingWorm, 2020-02-11:
Hi,
Thanks so much for the work you've put in - it's going to change my workflow. Simple brilliant.
I've only just gotten into 3D printing, so my knowledge is basic at best.
I've printed a few things in Cura with your profile (I have a FlashForge Pro).
However when I try and print this model:
https://www.thingiverse.com/thing:3126937
It doesn't seem to feed when I print to Octoprint in Cura.
The print head moves into position, but then the right extruder fails to flow and makes a thumping noise (like it's jammed or blocked). I manually fed the filament in a few times and confirmed it seems to be okay.
I print via Flashprint and it works okay with the same extruder.
I also saved the .gcode file and uploaded into via Octoprint directly - same effect.
What information can I provide to be meaningful?
I've experienced what a great workflow can be through your awesome work, it'd be shame to have to revert back to flashprint every so often.

Cheers,

Toylerrr, 2020-03-08:
Does it work without octoprint?

Also sorry for the late responce!

glietsch, 2020-05-14:
Hello, can you help me install the flashforge creator pro profile for cura please

closed 2020-06-09 by Toylerrr