Toylerrr/Flashforge-for-Cura #9: Extruder not extruding

shiruken1, 2021-10-19:
Hello and thank you for creating this. I've followed all the steps and can write out an x3g file, but the extruder won't actually put out filament. I remember some time ago something about the start code needing tweaking. Any advice? I tried to add you on Discord to no avail :(
[image]

[image]

Toylerrr, 2021-10-19:
Interesting, can you out put it to G-Code and then send it to me on discord or you can post it here whichever one you want.

After I stopped paying for discord nitro they changed my numbers Toylerrr#7035

gclloyd0, 2021-12-30:
Hi!  I would also like to thank you for creating this plug-in.  Was a resolution found to the problem mentioned in this thread?  I am experiencing the same issue trying to get Cura up and running for my FFCPs.  Running Cura 4.12.1 with the X3GWriter plug-in from Ghostkeeper (https://github.com/Ghostkeeper/X3GWriter)