Toylerrr/Flashforge-for-Cura #14: Im sorry, I have a creator pro, and when I put the nicely formatted files, it reads them, heats up right, then the issues start

Wilde-Bill, 2025-01-09:
My issue is that the bed levels to the correct position then goes all the way to the bottom of the printer, and tries to go down more with a buzzing noise, then the printer just starts printing in the corner  not in the middle where the design is.

Toylerrr, 2025-01-09:
This no longer works with the most recent version of cura, what version did you install this on? sounds like center origin is not checked in the printer setting in cura.

 I would recommend trying to use the Dreamer NX profile with x3g writer on the newest version of cura. They seem like they are basically the same.

Wilde-Bill, 2025-01-09:
jeez you responded fast
i fixed the center origin problem, but now my printer isn't doing anything at all. what should I do, do you have any ideas, cound it be start and finish g code in the custom fff printer I made on cura.

Wilde-Bill, 2025-01-09:
on top of that, the dreamer NX seems to have many different specs on it's page than the creator pro, a different build volume and only 1 print head mainly.

Wilde-Bill, 2025-01-22:
I tried Dreamer NX and it failed, I have tried a lot of things, I asked an AI, nothing works, and the ai finally recommended I ask people for help. Can you help?